mXlab/xenolalia #20: Run the OLED Mono using I2C instead of SPI

sofian, 2020-02-23:


sofian, 2020-02-24:
In the process I believe we fried our device. Ordering an alternative i2c screen.

sofian, 2020-03-03:
We had many problems with this. See the following discussion about running it from a RPi:
https://github.com/rm-hull/luma.oled/issues/276

tezgit, 2020-03-03:
I have been running two different I2C scanner sketches on Arduino Nano and on ESP32 with no success in detecting an address for the 128x128 I2C (only) display >> https://www.tinytronics.nl/shop/en/display/oled/1.5-inch-oled-display-128*128-pixels-white-i2c

After trying various HW I2C constructors with the U8g2lib I finally decided to give the SW I2C a try and managed to get the screen to work and run the page-buffer GraphicTest with this definition for 96x96 pixels 

U8G2_SSD1327_SEEED_96X96_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);  // Seeedstudio Grove OLED 96x96

[image]

This one actually worked too but the sketch seems to stop after displaying the first screen in full 128 x 128 resolution

U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);

[image]

and this is the backside of the display board

[image]

sofian, 2021-09-20:
Because of all the difficulties explained here we are falling back to work in SPI.

closed 2021-09-20 by sofian