mXlab/xenolalia #37: Print new design for screen support

etiennemontenegro, 2021-08-16:


sofian, 2021-08-18:
@etiennemontenegro My print failed again. Will try to adjust settings but I'm thinking maybe I should try printing vertically like @tezgit did originally.

etiennemontenegro, 2021-08-19:
Lets just print verticaly then. We're going to avoid another failure and have it ready for tommorow.  Strength is not a priority rn.

sofian, 2021-08-19:
Printed vertically. Material is very frail but I will try with different settings.

sofian, 2021-08-19:
Some notes to myself re: printing with PETG on Ender 3.
After printing a temperature tower I think it would be best to print with following settings:

First layer:
* Fan: 25%
* Hotend: 240
* Bed: 80

Rest:
* Fan: 80%
* Hotend (best): 240
* Bed: 80

Print with *brim* (not just a skirt).

Retraction: 6mm@45mm/s

Source: 
* https://old.reddit.com/r/ender3/comments/ec2i9j/how_to_calibrate_your_printers_esteps_and/
* https://www.thingiverse.com/thing:3912855

closed 2021-09-15 by sofian